steward-fu/gh_retrogame_emulator #29: fba freezing after crash

tackro, 2018-02-10:
after trying to play some games, fba crashes and need to reset, after reset I tried to play some games I had tested and confirmed to work, but fba crash, return to gmenu and freeze, so I have to reset too

tackro, 2018-02-10:
some more info:
I tried to replace default.cfg, frontend.cfg, fbasld.ini  and delete lastsel.cap (this file is corrupted and everytime I delete it reappears)
Ok, after this tried to play some games. Capcom ones run well, Neo-Geo don't recognize d-pad, and some other games like Armed Police Batrider freeze the emulator, return to gmenu and have to force reset. After this all games freeze the rs-97.

jutleys, 2018-02-10:
I have this same problem too games are running too slow aswell,

tackro, 2018-02-11:
windows repair usb should fix it, but everytime a game freeze needs to do this

steward-fu, 2018-02-16:
I also notice after fba failed to run some roms and return to gmenu2x, gmenu2x freezes.
I will check this issue

jutleys, 2018-02-16:
I have the correct rom set and the previews working the sound on a lot of games is bad and slowing it down aliensvspredator runs very nice but then aliens will slow and stutter with sound.

steward-fu, 2018-02-21:
since I try to reproduce it again, but I can reproduce it (freeze issue), can someone guide me how to reprofuce it ? thanks

jutleys, 2018-02-21:
try select+start button to exit may freeze fba.

tackro, 2018-02-21:
Try to run game Armes Police Battrider, FBA will freeze, after this try to L+R+select or start+select, nothing will happen, you will need to force reset device

closed 2018-03-13 by tackro