steward-fu/gh_retrogame_emulator #30: Opendingux splash screen

onthebridge, 2018-02-10:
Please add an opendingux splash screen when you switch on the handheld

steward-fu, 2018-02-10:
is it possible to provide it to me.
thanks

onthebridge, 2018-02-10:
Sure thing
[image]
[image]

jutleys, 2018-02-10:
Yes it would be great to get rid of the existing boot logo and replace it with this one

steward-fu, 2018-02-11:
Added opendingux splash on v20180213.

Since the default boot logo is controlled by kernel, it will take more time to patch it for new boot logo.
So, I will keep it.

steward-fu, 2018-02-16:
please close it

closed 2018-02-18 by onthebridge