steward-fu/gh_retrogame_emulator #34: Request to add this amazing wallpaper

jutleys, 2018-02-11:
[image]
Is it possible to add this wallpaper as it is very popular.
[image]

steward-fu, 2018-02-11:
Added as w1.png and w2.png under default theme on v20180213

jutleys, 2018-02-11:
[wallpapers.zip](https://github.com/steward-fu/gh_retrogame_emulator/files/1714613/wallpapers.zip)
Added wallpaper pack you may want to add some of these.

jutleys, 2018-02-11:
DekUI.zip added full theme files
Wallpapers.zip added many wallpapers.

steward-fu, 2018-02-16:
I will add

steward-fu, 2018-02-21:
ha ha, I overlook this issue, now, I am taking more time to add your great wallpaper and UI images.
Thanks for your great help :)

steward-fu, 2018-02-21:
Added in v20180313.
Amazing !
Thanks