steward-fu/gh_retrogame_emulator #44: [Port Request] Native games

ogbaba, 2018-02-12:
Hello, would it be possible to add native games such as 
- [ ] SuperTux
- [ ]  SuperTuxKart 
- [ ] Minetest
I don't really know if SuperTuxKart and Minetest could run though

ogbaba, 2018-02-13:
Here is a port of SuperTux for the GCW0 https://github.com/dmitrysmagin/supertux

steward-fu, 2018-02-16:
it sound good.
I will try to add it

ogbaba, 2018-02-18:
Ok so I got the version of supertux I mentionned running fine with sdlfix https://github.com/JackD83/sdlfix_rs-97

```
#!/bin/sh
LD_PRELOAD=./sdlfix.so ./supertux

```
although I had to compile with the -DNOSOUND flag to disable sound or it wouldn't work

steward-fu, 2018-02-20:
SuperTux have been ported to RetroGame with sound support.
It will be included in next release.