steward-fu/gh_retrogame_emulator #45: [Port Request] ONScripter

yunxiyinzhe, 2018-02-13:
Onscripter（Open Source Nscripter ）是一个用来解析NScripter脚本的第三方开源Galgame游戏引擎，可以在多种平台和设备上（Windows，Linux，FreeBSD，Android，iOS，Symbian……etc）上运行使用NScripter脚本编译的文字冒险类游戏。

这里是源码，基于SDL2：https://bitbucket.org/jh10001/onscripter-jh/overview

我看到makefile由针对Pandora和GP2X的移植，不清楚在Retro Game上移植是否可行。

感谢~

steward-fu, 2018-02-20:
請問一下，這個引擎可以玩哪些遊戲？
目前看來，應該是可以基於GP2X移植

yunxiyinzhe, 2018-02-20:
https://wapbaike.baidu.com/item/ONScripter/8936792
应该有很多gal game可以玩。

yunxiyinzhe, 2018-02-26:
原版的代码在这里可以找到：http://onscripter.osdn.jp/onscripter.html
是基于SDL1版本的，那个SDL2版本的代码并没有针对GP2X进行过修改，编译不过，Makefile也不太对。虽然改改也能编过。。

steward-fu, 2018-02-26:
OnScritor-Jh has been ported to RetroGame and will be included in next release:
package:
https://drive.google.com/file/d/1NtcFTlZiQ7TcZjscCLyIHLOl5neg6abd/view?usp=sharing

steward-fu, 2018-03-02:
由於速度及常常發生Crash問題, 因此下版釋出將不會包含, 不好意思 ！