steward-fu/gh_retrogame_emulator #49: Extend the game partition upgrade error

jutleys, 2018-02-13:
I wanted to extend the game partition higher to 16gb when you try to run the upgrade official firmware fails to mount and upgrade fails. Could you try and fix this It might be looking for the original file size of the game partition because its been changed to 16gb or more.

steward-fu, 2018-02-13:
upgrade firmware just do mount and umount.
it will not effect the partition size.
could you provide the sha1 value of your upgrade.ext3 file.

jaZzKCS, 2018-02-13:
Is it possible to extend the game partition after doing a full flash installation?
As in: First flash the entire .img version and then extend the partition manually? Will the device be bothered?

steward-fu, 2018-02-13:
sure, I will consider it.
but I want to know what's the means of "Extend the game partition upgrade error" ?

jutleys, 2018-02-13:
I am trying to install on 128gb card when i upgrade it keeps saying mount failed

jutleys, 2018-02-13:
I have not touched any partitions yet only default ones.

steward-fu, 2018-02-13:
do you have installed beta os on this 128gb sdcard ?

jutleys, 2018-02-13:
Sorry Steward its my fault i was extracting upgrade files to game folder instead of the roms folder problem solved.

jutleys, 2018-02-13:
Thank you for your help.

steward-fu, 2018-02-13:
btw, do you have pap k3 (7 days... all of default pre-installed games), please provide to me.
I would like to porting even reverse engineering to retro game.
thanks

steward-fu, 2018-02-13:
good. hope you enjoy !

jutleys, 2018-02-13:
I have extended the game partition to 16gb and 100gb for roms partition i like this setup.

closed 2018-02-13 by jutleys

jutleys, 2018-02-13:
I have the original  .img file here
https://drive.google.com/file/d/0B0wz_n5he9GpQW96allVTUJNSnM/view

jutleys, 2018-02-13:
I have the original  .img file here https://drive.google.com/file/d/0B0wz_
n5he9GpQW96allVTUJNSnM/view

steward-fu, 2018-02-13:
thanks :)
I will try to port to RetroGame device

jutleys, 2018-02-13:
Cool

steward-fu, 2018-02-13:
I cannot see any pre-installed game like 7-days.
do you have seen them on pap k3 ?

jutleys, 2018-02-13:
I never saw that game all my roms were preinstalled english maybe you need
someone with papkiii plus with the chinese rom preinstalled that could have
the game on there.

steward-fu, 2018-02-13:
got it, thanks