steward-fu/gh_retrogame_emulator #50: Can't mount game folder since upgrade

jutleys, 2018-02-13:
When i connect usb now since the upgrade i can see the roms folder but not the game folder.

jutleys, 2018-02-13:
Sometimes it shows sometimes it does not usb mounting as better behaved on last version.

steward-fu, 2018-02-13:
Game folder has been removed from mount list in official version because it is very rare to change it, right ?

jutleys, 2018-02-13:
Ok thanks for confirming.

jutleys, 2018-02-13:
It feels more unstable for usb mounting my roms folder sometimes does not show up but on the last beta it always shows when i click usb icon.

tackro, 2018-02-13:
Another reason because it's better have partition Game mounted is that fbasdl bug that corrupt lastsession.cp everytime that a game freeze

steward-fu, 2018-02-16:
I will try to add into mount list but since gmenu2x also is in this partition, I need to check whether gmenu2x can work well when mount to pc