steward-fu/gh_retrogame_emulator #53: Request to port 7 days dingoo a320 version

jutleys, 2018-02-13:
Here is 7 days dingoo a320 version request to port it to tetrogame

[7days_201408.zip](https://github.com/steward-fu/gh_retrogame_emulator/files/1720822/7days_201408.zip)

steward-fu, 2018-02-13:
you are hero. thanks :)

steward-fu, 2018-02-13:
this file is copied from pap k3 ?

2018-02-14 1:06 GMT+08:00 Steward Fu <steward.fu@gmail.com>:

steward-fu, 2018-02-13:
as I known, the file format belongs to dingoo a320.
if this app can run on pap k3, it is like rtos for pap k3, not dingux.

jutleys, 2018-02-13:
I am not sure about the papkiii plus version i have not seen it.

panshuai0, 2018-02-14:
7 nights, papk2 should bring their own.

steward-fu, 2018-02-16:
since original dingoo run rtos, we need rtos emulator for this.
put it into long term list