steward-fu/gh_retrogame_emulator #54: PocketSNES - Video and Audio Play Slow

kaplanfx, 2018-02-13:
Games play slowly on PocketSNES (On 1.0 release of the OS). Frame skip doesn't help, it just drops frames rather than speeding up the play to the correct speed.

jaZzKCS, 2018-02-13:
Generally all of the ones I tried (around 20) run all at the same speed which feels around 80%.
The FPS indicator of the emulator reports 41-44/50 in all games.

kaplanfx, 2018-02-13:
The weird thing is that frame skipping makes it worse. I set frame skipping to 1 and I get 22 frames instead of 44 but the slow speed persists.

tackro, 2018-02-13:
@steward-fu did you saw Lion .SIM version for dingoo native of a snes emulator? it's called DingooSNES and uses snes9x143 as base. I'll try to send you the source code. I think that it has greater performance

steward-fu, 2018-02-14:
I got it, thanks you.
I will check it soon. :)

jaZzKCS, 2018-02-14:
I tried an old SNES9x4D version from 2010 and it had slightly more performance than pocketsnes. Maybe the newest version has better performance?

steward-fu, 2018-02-16:
@JaZzKCS new version ? Snes9x4d ? 
for pocketdsnes fps skip issue, I need more time to check it.

jaZzKCS, 2018-02-16:
I have not seen a newer version of 4D. Maybe the 2010.10.10 is the most recent?