steward-fu/gh_retrogame_emulator #57: Feature Request (duelboot) from external memory card

jutleys, 2018-02-14:
Request if its possible to boot original firmware from external memory card with a button press on boot which will boot from external card.No press will boot from internal card into gmenu2x custom firmware i think this could be a amazing featurw if it can be done Steward.

jutleys, 2018-02-14:
This way the user can play original firmware emulators or new gmenu2x emulators.

steward-fu, 2018-02-16:
sure, but it might need new buiboot or reverse to patch uboot loader.
need some time for checking it

jutleys, 2018-02-16:
That sounds good Steward i think this will be a popular option as it gives people a way to go back and play the games that came with their devices. So this way they will have the best of both worlds.

panshuai0, 2018-02-16:
Yes, I hope to open dual-system startup！The original system can run the Chinese name game，Game rom can be placed anywhere，Simple and practical system~