steward-fu/gh_retrogame_emulator #59: [pcxs4all] X-button (Triangle) (upper button) not mapped / not working.

jaZzKCS, 2018-02-14:
The x-button on the device (upper button), which corresponds to the PSX "triangle"-button, is not mapped it seems, resulting in not working in emulation. There is no option to change the controller settings in the emulator menu, therefore I open up this ticket.

jaZzKCS, 2018-02-15:
Can someone direct me to the config file I need to change if that is possible? A non-working triangle button kind of hinders me to test further.

steward-fu, 2018-02-16:
which game do you play in pcsx4all ?

jaZzKCS, 2018-02-16:
I tested out FFVII and Tekken 3 so far and noticed the triangle button not working on both games, therefore concluded it isnt mapped.

steward-fu, 2018-02-16:
after check with Raiden rom, X button is work.
could you kindly help to check whether it is different withwithout HLE bios

jaZzKCS, 2018-02-16:
Sorry, I can't. It starts with HLE enabled. When going into settings and turning it off, exiting the emulator and restart, it didnt save the setting (it's on again). And when changing the setting and not exiting, but going back to emulation, it crashes. I cannot turn it off, it wont save it :(

jaZzKCS, 2018-02-18:
Please close this issue, it turned out my unit had a hardware fault (capacitor missing for "x"-button, they are very small and it's easy overlook that one is missing.)

closed 2018-02-19 by steward-fu