steward-fu/gh_retrogame_emulator #61: snes9x4d

gigiwotou, 2018-02-14:
用新的模拟器玩快打旋风帧数稳定40/60.
玩塞尔达中文版，开场可以60/60.刚出房子外部场景元素丢失。40/60.
http://img0.ph.126.net/lcwvffXqNlpBO5WBF1wt7A==/5717545126830789987.jpg

gigiwotou, 2018-02-14:
打开transparency可以显示正常。帧数28/60

jutleys, 2018-02-14:
Does not save settings.

steward-fu, 2018-02-16:
所以問題是模擬器效能問題嗎？

steward-fu, 2018-02-16:
@juleys  for setting issue, could you submit an issue. the issue for gigiwotou seems to be performance issue for snex9x4d

jaZzKCS, 2018-02-16:
Have tested SNES9X4D v20101010 and can confirm all standard games (no Mode7 or SuperFX2) running at 50/50 and 60/60 respectively.

Best results are had with the following settings (have to be set for each game individually for now):

-Transparency : on
-Fullscreen: True
-Frameskip: Either 50/1 or auto (depends on game)

Note: Parallax heavy games, as well as Mode7-heavy games do still have performance issues.

Albeit the vast majority of games I tried run very close to if not at full speed. If they don't for you, try switching between framskeip 50/1 (off) and auto settings.

jutleys, 2018-02-16:
I can only see 60/1

jaZzKCS, 2018-02-27:
@jutleys That makes sense if you have a NTSC rom loaded.

jaZzKCS, 2018-02-27:
Only games without parallax, mode7 or any other enhancements are running at full speeds at the moment.