steward-fu/gh_retrogame_emulator #62: OhBoy doesn't support RTC (real time clock)

literallywhy, 2018-02-14:
There's no support for the RTC in the Pokemon games on the GBC. Time doesn't sync to the system clock on the RS-97 (if there is one)

jaZzKCS, 2018-02-14:
There is none. The RS-97 doesnt have a real time clock. Clock starts from 01/01/1980 or 01/01/1970 respectively once the device is started.

literallywhy, 2018-02-14:
Can we implement a way to change the time on the OS side then? It should be keeping time while the battery is inserted/charged like how the PSP or DS do it right?

Also, if thats the case would ohboy just go ahead and use the system time or would the actual emulator need that feature added? I imagine the former is easier to do than the latter.

literallywhy, 2018-02-14:
I don't know if the SoC has an RTC, but here's what I'm talking about:
https://www.kernel.org/doc/Documentation/rtc.txt

New portable "RTC Class" drivers:  /dev/rtcN
--------------------------------------------

Because Linux supports many non-ACPI and non-PC platforms, some of which
have more than one RTC style clock, it needed a more portable solution
than expecting a single battery-backed MC146818 clone on every system.
Accordingly, a new "RTC Class" framework has been defined.  It offers
three different userspace interfaces:

    *	/dev/rtcN ... much the same as the older /dev/rtc interface

    *	/sys/class/rtc/rtcN ... sysfs attributes support readonly
	access to some RTC attributes.

    *	/proc/driver/rtc ... the system clock RTC may expose itself
	using a procfs interface. If there is no RTC for the system clock,
	rtc0 is used by default. More information is (currently) shown
	here than through sysfs.

The RTC Class framework supports a wide variety of RTCs, ranging from those
integrated into embeddable system-on-chip (SOC) processors to discrete chips
using I2C, SPI, or some other bus to communicate with the host CPU.  There's
even support for PC-style RTCs ... including the features exposed on newer PCs
through ACPI.

The new framework also removes the "one RTC per system" restriction.  For
example, maybe the low-power battery-backed RTC is a discrete I2C chip, but
a high functionality RTC is integrated into the SOC.  That system might read
the system clock from the discrete RTC, but use the integrated one for all
other tasks, because of its greater functionality.

Check out tools/testing/selftests/timers/rtctest.c for an example usage of the
ioctl interface.

literallywhy, 2018-02-14:
Yup, the JZ4760 does support RTC:
Check section 1.2.6
https://www.rockbox.org/wiki/pub/Main/IngenicJz47xx/JZ4760_pm.pdf

literallywhy, 2018-02-16:
I can confirm that OhBoy does keep track of time while the emulator is running. For example in Pokemon Crystal I set the time to 10pm and it was advancing with the game (probably synced to the emulated CPU cycles). But after you exit the emulator and reboot the device the time stays the same, rather than syncing with the system clock and by comparing the game time with the system time and *simulating* having a battery-backed timekeeping module. This breaks gen 2 and 3 of Pokemon and other games that rely on the RTC like Boktai etc. 

The simple fix is a PSP-like solution where the system keeps time as long as the main battery is charged and inserted. If either of that changes the PSP prompts you to entire the time and date on boot. The PDF I posted above confirms that is a feasible solution. A suspend function would also be appreciated by many users I'm sure since powering the device on and off for every use gets annoying.

literallywhy, 2018-02-16:
And to elaborate on your second question ohboy does do that, but there's no steady system clock for it to sync with as I stated above.

literallywhy, 2018-02-18:
I just noticed that when you make a save state in ohboy, it uses the time and date counting from 00:00 1980, meaning opendingux counts time using the linux epoch like normal. 
So a simple fix to the RTC problem without having to add any complicated modifications to the emulators or the OS would consist of two parts:

1. Prompting the user to set the date/time at boot
2. A better suspend/standby system like how the PSP works or closing the lid on a laptop screen, that way the system will still keep time while not in use in an easy to implement way.

steward-fu, 2018-02-19:
so, if gmenu2x support RTC time (read from RTC register), maybe, it can fix your problem.
I will consider to implement it.

literallywhy, 2018-02-19:
Yep. Thanks!

Soviet5lo, 2018-03-09:
When/If RTC support gets added to this device, you could always use dgclock to set the time and date.  However, I'm not exactly sure if the kernel the RS-97 is using currently supports RTC.

http://clogging.web.fc2.com/dingoo/ (Source code included in its archive)