steward-fu/gh_retrogame_emulator #63: [Feature Request] Port of several emulators

onthebridge, 2018-02-14:
Amstrad CPC https://boards.dingoonity.org/gcw-releases/caprice32-(amstrad-cpc)/
Amiga 500 https://boards.dingoonity.org/gcw-releases/uae4all-release-1/
Spectrum 48/128k https://boards.dingoonity.org/gcw-releases/unreal-speccy-portable-0-0-59/

tackro, 2018-02-15:
slaanesh is a great person, if someone contact his i think he will help with this, aside, their emulator are good and optimized

steward-fu, 2018-02-16:
for xmame, I have ported but still need to fix some things like failed to load *.bin file.
for other emulators, I will add into porting list.

steward-fu, 2018-02-25:
xmame can run on RetroGame but with very low fps and sound lag.
I will put it into low priority list.
Amstrad CPC and Amiga 500 will be take and port soon.

steward-fu, 2018-02-25:
Amstrad CPC has been ported to RetroGame and will  be included in next release.
package (put under /mnt/game/cap32/):
https://drive.google.com/file/d/1COaCJ-IHUdR4pDEucniTaC-06U3jdkmH/view?usp=sharing

steward-fu, 2018-02-25:
Amiga 500 has been ported to RetroGame and will be included in next release.
package (put under /mnt/game/uae4all):
https://drive.google.com/file/d/1Ft7wSwutFdGG-OnuLPqs0w0FZJSIlW3W/view?usp=sharing

onthebridge, 2018-02-25:
Amazing Steward, thank you very much :D

closed 2018-02-25 by onthebridge