steward-fu/gh_retrogame_emulator #64: Jump to next page by pressing left/right button

tri2510, 2018-02-15:
Could you add the feature so that we can jump through page of games quickly by pressing left/right button instead of continuously pressing up/down button.

kaplanfx, 2018-02-15:
Depending on where you are talking about in the system, the L and R buttons might act as page up / page down.

jaZzKCS, 2018-02-15:
Jumping quickly through the pages is already possible with L/R buttons.

But when having a full rom set present this is actually not enough in my opinion.
Especially with the new theme that "lags" when browsing through many files this takes very long.

Could I suggest adding also the function to select a letter from a list to quickly jump to the files starting with that letter?

@tri2510 Could you change your topic to smth like "Ability to select letter for browsing roms" ?

steward-fu, 2018-02-16:
please change the title.
I will add letter into gmenu2x for each emulator.
it will also need refresh function for this function.

tri2510, 2018-02-16:
@jaZzKCS I changed the title as your suggestion.
@steward-fu  thanks for the work 👍