steward-fu/gh_retrogame_emulator #66: [Port request] uNGP - Neo Geo Pocket Emulator

tackro, 2018-02-15:
here's the code:
https://code.google.com/archive/p/developia/source

tackro, 2018-02-15:
RACEOD is another emulator, here's the code
http://dingoo.a320.free.fr/downloads/emuls_opendingux/neogeo_pocket/race-od_v10_src.zip

steward-fu, 2018-02-16:
thanks for your information, I will port this soon

tackro, 2018-02-16:
@steward-fu I marked you on dingoonity, I tried to run RACE-OD with sdlfix.so and emulator runned but no image, but can hear the sounds of game. When execute without sdlfix it gave double screen (30fps on which screen)

steward-fu, 2018-02-16:
for sdlfix, since I don't know what issue causes according to the description. so, you can use gdb to debug this issue
https://github.com/steward-fu/gh_retrogame_dev

but it needs uart debug port for debugging

steward-fu, 2018-02-16:
by the way, I will also fix double screen issues in sdl library soon because fix each emulators are not good idea.

steward-fu, 2018-02-17:
1. race-od have been posted on dingoonity forum for testing and it will be added into next release list.
2. finally, I decide to give up fix double screen in lib sdl because it is more difficult.
if you can shared the sdlfix source code, I can help to check it.
thanks

steward-fu, 2018-02-18:
kindly help to check race-od emulator again and feedback issue:
https://drive.google.com/file/d/1nD4zXaFb72uaPKwZd3jse0yynS_0vu5i/view?usp=sharing

steward-fu, 2018-02-18:
Since uNGP source code is not available anymore, ignore uNGP emulator.

tackro, 2018-02-18:
Framerate now is better on native res (around 47-50fps) but when change rom, the menu still on screen (only when select native resolution). On fullscreen still the same, 30fps on all games.

steward-fu, 2018-02-19:
bugfix: skin redraw issue
https://drive.google.com/file/d/14iRbuBpEn7Alvvz8j3hRPutLHHAf4IDv/view?usp=sharing

for performance issue, I will reply in another issue you submitted.
thanks

steward-fu, 2018-02-19:
please kindly help to provide the 32x32 png icon for gmenu2x with background transparency.

tackro, 2018-02-21:
Emulator aspect ratio change it's fixed, thanks.
There's only a strange horizontal line on top of screen but its a minor.
Framerate is ok on aspect ratio, 50fps, playable.

steward-fu, 2018-02-23:
this is original picture for skin:
https://drive.google.com/file/d/1XxF_lO9E1NqImaUXGbJt62v1STRAqIYm/view?usp=sharing

if you can provide new one, I can help to update it,
thanks

steward-fu, 2018-02-23:
Fix: extension name issue
https://drive.google.com/file/d/1K6_0E37H84vP1G5jfLfA7hKTch7GbMY-/view?usp=sharing

closed 2018-03-13 by tackro