steward-fu/gh_retrogame_emulator #72: Start + Select to get to menu

FantasticMind, 2018-02-16:
It would be great to have start+select for every emulator.  It can get confusing at times as some emulators have a different button combo to get to the menu.

jaZzKCS, 2018-02-16:
It took me quite a while to find out the combination you need to press for pcsx4all is select followed by start, not both at the same time. I agree, a unified combo would be nice, if realizable.

panshuai0, 2018-02-16:
Guys, my advice is ： Power button to restart the device, the top key for the unified simulator menu, cross keys up and down any of ABXY up and down any combination to adjust the brightness and adjust the sound
电源键为设备重启，顶部键为统一模拟器菜单，十字键上下左右任意+ABXY任意组合为调整亮度和调整声音

steward-fu, 2018-02-19:
this feature is great but it will be put into low priority list because it need to change each emulator.
thanks