steward-fu/gh_retrogame_emulator #73: Request port (gngeo neogeo emulator)

jutleys, 2018-02-16:
Request port (gngeo neogeo emulator)

https://boards.dingoonity.org/gcw-releases/gngeo-port-(neogeo-emulator)/

steward-fu, 2018-02-17:
which version do you want v0.7 or v0.8 ?

jutleys, 2018-02-17:
v0.8

steward-fu, 2018-02-18:
since v0.7 and v0.8 have some issues on compiler, so, I used pi-gngeo-v0.85 for porting.
please help to test it and feedback issue:
https://drive.google.com/file/d/1qHMV742nVjN4-B3Zah5gv6pR4l3kVsZR/view?usp=sharing

since I decide not to post any develop progress and beta testing emulator on forum (teiba and dingoonity) because some people's expectation is too high (this is not good for me), so, if you would like to help test with another people, please feel free share it.
thanks for your help.

jutleys, 2018-02-18:
> B3Zah5gv6pR4l3kVsZR/view?usp=sharing

OK np

steward-fu, 2018-02-18:
sorry, please ignore this version because there is a key mapping issue.

steward-fu, 2018-02-18:
I will update it soon

jutleys, 2018-02-18:
OK no problem

steward-fu, 2018-02-19:
kindly help to test it:
https://drive.google.com/file/d/1cGQEGUfW-Vge6czYCvOlI86CHw9A0V5c/view?usp=sharing

since "effect" is not implemented yet, is it need to implement ?

steward-fu, 2018-02-19:
please kindly help to provide the 32x32 png icon for gmenu2x with background transparency.

jutleys, 2018-02-19:
I am going to upload the whole of my icon set for you has every console in
it.

jutleys, 2018-02-19:
Ok Steward icons uploaded.

steward-fu, 2018-02-19:
can I share these icons on this github (I mean I will add these icons into this repository) ?

jutleys, 2018-02-19:
sure and how do you install gngeo where do skins and rom folder go

steward-fu, 2018-02-19:
1. unzip 
2. create linkapp (gngeo)
3. launch this app (gngeo has itself GUI for loading rom)

I forget to test skins...

jutleys, 2018-02-19:
the error i am getting is 

Gngeo couldnt init burningf:
Rom/mmt/int_sd/neogeo/burningf.zip not found
Maybe the romset youre using is too old.

steward-fu, 2018-02-19:
which rom do you test ?

jutleys, 2018-02-19:
all of them none are working for me they are all zipped.

jutleys, 2018-02-19:
Does data.zip and skin.zip have to be unzipped to?

steward-fu, 2018-02-19:
can you pack all of your neogeo folder to me, I will check it.
I only test kof98.zip, it can work well

jutleys, 2018-02-19:
I extracted the folder to game and run dge file it may be my romset.

jutleys, 2018-02-20:
Only game working for me is (3 Count Bout) when i try any other game just same error blank screen then above error.

jutleys, 2018-02-20:
Amazing speed though 60fps steady very smoothe.

jutleys, 2018-02-20:
Which set is working with this emulator.

steward-fu, 2018-02-20:
since this porting is from pi-gngeo v0.85, so, I don't know which set is working for this emulator.
but I think I need to remove "effect" or to implement it in this emulator

jutleys, 2018-02-20:
very good speed prolly the best so far need to find out the set which works i will check.

jutleys, 2018-02-20:
https://github.com/ymartel06/GnGeo-Pi/wiki

jutleys, 2018-02-20:
set 0.138.

steward-fu, 2018-02-20:
haha, thansk for your information.