steward-fu/gh_retrogame_emulator #81: Device not booting after pluggin in USB-power at low battery

jaZzKCS, 2018-02-18:
In the original Chinese firmware, the same would happen, this blinking symbol would get displayed. When you then plug in a charging cable, it starts charging and continues to boot to the desktop.

With the custom firmware, the last step doesnt happen, hence you have to attach a loading cable and then press the reset button.

steward-fu, 2018-02-19:
do you mean this device will not reboot into desktop when in poweroff status ?

jaZzKCS, 2018-02-19:
What I mean is:

When you start your device and its low on battery, instead of booting normally it will show the blinking battery splashscreen.

Then, if you plug in a cable it would start charging, the blinking splashscreen goes away and it continues to boot. 

With your custom firmware this doesnt happen. You put in the charging cable but it will continue to endlessly show the blinking battery symbol. You cant shut it down, and you cannot boot either. The only thing you can then do is to press reset button. 

Wit

jaZzKCS, 2018-02-20:
Huh, this is weird. Today when I wanted to start it up and got the low power splashscreen, it actually rebooted once the USB cable was attached, which it didn't do before on cfw, only on official fw.

This seems to be sporadic behaviour

steward-fu, 2018-02-21:
basically, all of usb issues are related to kernel/hardware issue, not related to gmenu2x, hope you understand and kindly close this issue. thanks

jaZzKCS, 2018-02-21:
"basically, all of usb issues are related to kernel/hardware issue, not related to gmenu2x, hope you understand and kindly close this issue. thanks"

@steward-fu Im sorry, but I cannot close issues (even ones I created), only you can! :P

steward-fu, 2018-02-21:
ok, thanks :)

closed 2018-02-21 by steward-fu