steward-fu/gh_retrogame_emulator #85: MULTIMEDIA APPS FOR RS-97

marianovl123, 2018-02-20:
Hi steward.

Should be possible port anykind multimedia app (to play .mp3, see videos, read books, etc) for RS-97 ??

Perhaps in a next release ??

Thx. in advance.

Regards.

onthebridge, 2018-02-20:
For example ...
Bard Storyteller http://festvox.org/bard/
Oldplay https://boards.dingoonity.org/gcw-releases/oldplay-for-opendingux/
Gmu Music Player https://boards.dingoonity.org/gcw-releases/gmu-music-player-for-gcw-zero/
FFplay Media Player https://boards.dingoonity.org/gcw-releases/ffplay-media-player/
CBZviewSS comic reader https://boards.dingoonity.org/gcw-releases/cbzviewss-comic-book-reader-(-cbz-viewer-for-small-screens)

marianovl123, 2018-02-21:
Many thx onthebridge.

I’ll wait for compilation of these apps, even the inclusión of this in a new CFW reléase.

Thanks in advance.

Regards.

NOTE: I still remember little kind of knowledge about compiling C since University days. Is there any kind of tutorial (compilator versión ¿? Libs ¿? Gcc flag’s to conver into .dge for dingoo, etc.) to try convert those apps by myself ¿?

steward-fu, 2018-02-21:
I will check it and evaluate whether they can be put into next release, thanks

steward-fu, 2018-02-23:
GMU has been ported to RetroGame and it will be included in next release.
package (put under /mnt/game/gmu/):
https://drive.google.com/file/d/1uCmoZiTiUs3mgJN4cS0Ls0KpddezMdhM/view?usp=sharing

https://steward-fu.github.io/website/test/25.jpg

steward-fu, 2018-02-24:
Bard has been ported to RetroGame and it will be included in next release.
package (put under /mnt/game/bard/):
https://drive.google.com/file/d/1aqzI96bnqUwfxPUWqq46YPz0pL3nwdXt/view?usp=sharing