steward-fu/gh_retrogame_emulator #86: Screen Lock - Long Press Power Off

FantasticMind, 2018-02-22:
So I tried to start up kirby on the gba emulator.  It started, but the locked on start screen.  I couldn't press any buttons to get back to the menu or power off.  I was locked.  

I had to unscrew and take the battery out to power off.  I tested this to see if it was random.  It happened again.  It could be a bad rom.  Not sure.  

The situation is though, it may happen where the screen locks and you need to long press the power button to power off the device to restart.

Is it possible to code a long press (3 seconds) to power off the device?

steward-fu, 2018-02-22:
do you see reset button on the bottom side ?
as I known for your issue, gba emulator crash and then back to GMenu2X but SDL resource might not be freed properly by gba emulator.
so, gmenu2x cannot be launched again.

in such case, you can only reset your device.

add poweroff feature is good idea, but since lack of kernel support, I cannot install any kind of driver on this device.
so, poweroff cannot be implemented so far.

jaZzKCS, 2018-02-22:
This also happens a lot when you start your device with low battery resulting in you being trapped with the low battery splashscreen even if you attach USB power cable. The only way to restart then is the same as when an emulator crashes. You have to press the reset-button (preferably with a small screw driver)

There is no need to plug out the battery.

Until the emulators are worked over in terms of compatibility, expect some crashes here and there, especially with GBA emulation. Steward is still in the process of adding emulators and features and the phase of optimizing aforementioned emulators will come at a later point.

FantasticMind, 2018-02-22:
When I hear reset, a lot of things come to mind.  But basically a reset on this image just means reboot?

Nothing will be effected?

jaZzKCS, 2018-02-22:
Yes, hitting the reset-button  at the bottom of the device will not erase settings, it will only force a restart of the device.