steward-fu/gh_retrogame_emulator #90: czdoom issue

jutleys, 2018-02-26:
when clicking setup in option resets back to home screen

steward-fu, 2018-02-26:
so, quickly selection to find out an issue, ha ha

jutleys, 2018-02-26:
I was waiting for doom long time how do you load doom1.wad  get all others
to load but not that one.

steward-fu, 2018-02-26:
do you load wad file from gmenu2x ?

steward-fu, 2018-02-26:
root cuase:
Couldn't load MIDI from /tmp/prboom-music-cLEHPu: Unrecognized music format

it seems the music format not support by current system.
I will check it soon.
Thanks

steward-fu, 2018-02-26:
root cause for back to desktop when clocking on setup:
# W_GetNumForName: M_CHAT not found
I_ShutdownMusic: removing /tmp/prboom-music-cLEHPu

steward-fu, 2018-02-26:
doom1.wad issue has been fixed in next release (v20180313) by recompiled sdl-mixer with midi support.

jutleys, 2018-02-26:
Great stuff Steward.

steward-fu, 2018-02-28:
For SETU crash issue, after tested on PC with doom1.wad, dom2.wad, it is same as RetroGame.
Is it possible the *.wad corrupted ?

jutleys, 2018-02-28:
Wads are fine as i used them on another handheld `gamegadget and are all working.

steward-fu, 2018-03-01:
Finally, I decide using chocolate-doom to replace czdoom because czdoom seems a lot of issues need to be fixed.
Kindly help to test chocolate-doom:
https://drive.google.com/file/d/1vHeN10lp2TDMgNb2Xp8k2CcCIjQxd6ml/view?usp=sharing

thanks

jutleys, 2018-03-01:
> 6ml/view?usp=sharing

Ok Thanks Steward.

jutleys, 2018-03-01:
> xd6ml/view?usp=sharing

Okay well it runs very good Steward is there any way to get rid of the
black borders on top and bottom? i tried all the wad files and everytime i
change the wads with the file browser it only starts doom2 default am i
doing something wrong?

jutleys, 2018-03-01:
And maybe change the button configuration better use 

Button Configuration

holder buttons for sidestep ,
B for fire,
Y to open doors,
A run,
X change weapon

jutleys, 2018-03-01:
But i found no ther bugs apart from the above nice port.

steward-fu, 2018-03-02:
Fixed:
1. stretch to fullscreen
2. swap button

linkapp:
  params: -iwad $f

https://drive.google.com/file/d/1H2lW6Xng_HJc6ci2_i-YE_j5Ltxs3HGo/view?usp=sharing

jutleys, 2018-03-02:
> j5Ltxs3HGo/view?usp=sharing

Thanks testing now Steward.

jutleys, 2018-03-02:
Its working perfect now i can change wads and they all work key binding is
better but the trigger shoulder buttons still are changing weapon if you
can get them to do (sidesteps left and right) that would be great Steward
great work.No other bugs to report.

steward-fu, 2018-03-02:
try this for trigger button:
https://drive.google.com/file/d/1yscD7Egk5YkDsLZDvFc1EgdCEPbgQmJd/view?usp=sharing

jutleys, 2018-03-02:
> try this for trigger button:
> https://drive.google.com/file/d/1yscD7Egk5YkDsLZDvFc1EgdCEPbgQ
> mJd/view?usp=sharing

Its perfect now thank you ready for release i think great job.

steward-fu, 2018-03-02:
I will use chocolate-doom to replace czdoom in next release.
Do you think czdoom that is still worthy to keep or not ?

jutleys, 2018-03-02:
Yes because it still runs good gives people a choice.

steward-fu, 2018-03-02:
Got it, thanks :)