steward-fu/gh_retrogame_emulator #97: [Emulators Request] NP2 (Nec PC-88/98) and OpenMSX

tackro, 2018-03-02:
Here the sources:
Neko Project 2 (NP2)
https://dl.dropbox.com/u/525714/Open-Source%20Emulators/dingux-np2-src.zip

openMSX:
Opk: https://github.com/openMSX/openMSX/releases/download/RELEASE_0_14_0/openmsx-0.14.0.opk
Source: (0.9.1) http://dingoo.a320.free.fr/downloads/emuls_opendingux/ordinateur/openmsx-0.9.1.tar.gz
Source: (0.14) https://github.com/openMSX/openMSX/releases/download/RELEASE_0_14_0/openmsx-0.14.0.tar.gz
Github: https://github.com/openMSX/openMSX

jaZzKCS, 2018-03-02:
OpenMSX would be awesome. Looking forward to be able to play the original La-Mulana on it :)

steward-fu, 2018-03-04:
OpenMSX and NP2 have been ported into RetroGame and will be included in next release.
But since I am not familiar with 2 emulators, so, I just do simple  testing.
Please kindly help to test them in next release.
Thanks

tackro, 2018-03-05:
of course

closed 2018-03-05 by tackro