steward-fu/gh_retrogame_emulator #98: [pcsx4all[ Playstation emulation optimization

jaZzKCS, 2018-03-02:
This is just a placeholder/reminder for @steward-fu regarding to have a look upon PSX emualtion optimization, whenever there is time.

pcsx4all seems to be far from optimized and the performance is all over the place, depending on which game you try.

onthebridge, 2018-03-02:
I think this emulator is still in development by its original creators so it might be worth it to ask Dmitry Smagin about it and it they are going to support the rs97

jaZzKCS, 2018-03-02:
Yes, this is just a reminder for him to regularly check for a new version of this as it's updated rather frequently. :)

steward-fu, 2018-03-03:
reply from Senor Quack:
Unless there's a 2nd CPU core, I don't think multi-threaded drawing will not help.  One thing that will soon help gpu_unai speed is specialized optimizations for sprite drawing 4 pixels-at-a-time. Another thing that will help is cut-down GTE functions that skip computing flags for most things.  We actually now have custom MIPS MXU assembly functions for all GTE functions, but they're MXU rev2-only, so don't apply to jz4760, only jz4770. MXU rev1 is very limited and nearly useless.  Actually, we have MXU rev2 assembly replacements for all gpu_unai rendering and also audio, as well as IPU scaling support in the works.  It's slowly coming together.

steward-fu, 2018-03-03:
since jz4760 is with IPU (image process unit) feature, it is very powerful.
I will take more time to adopte IPU into pcsx4all.
Just keep this issue open in order to reminder me for this task.
Thanks