steward-fu/gh_retrogame_emulator #104: [Port Request] Gambatte (GB/GBC Emulator)

Soviet5lo, 2018-03-04:
Its a more accurate emulator emulator than Gnuboy/Ohboy, and it would be nice to have the option to pick between the two.  It's been ported to the A320 and GCW Zero, so I'm sure it can run on the RS-97.

Source code for upstream: https://github.com/sinamas/gambatte

Source code for Opendingux port: https://github.com/Surkow/gambatte-opendingux (original repo is here https://code.google.com/archive/p/gambatte-opendingux/)

steward-fu, 2018-03-04:
Gamebatte has been ported to RetroGame and will be included in next release.

Soviet5lo, 2018-03-04:
Thanks for the quick port.  Tried it out on my RS-97 and games run at full speed and look great.  However, I've gotten a crash + gmenu2x freeze whenever I exit the emulator, no buttons respond but I can change the brightness.  I'm not sure if this is due to the porting process or if it's a bug from the A320/GZW builds.

Soviet5lo, 2018-03-13:
Closing since its been ported and the issue I had doesn't exist in the new cfw.

closed 2018-03-13 by Soviet5lo