steward-fu/gh_retrogame_emulator #123: v1.2 firmware issues

jutleys, 2018-03-15:
TV out - works fine on the main menu, but trying to run an emulator makes the device freeze. Have not tried all emulators. Have not tried any games.

O2xIV - image viewer - distorts image when zoomed in and panning to the right only. Panning up, down, left, does not distort the image.

FCEUX - all video scalers don't display correctly except original, and aspect.

Atari 5200 - Games load into some weird blue screen where everything is distorted. Can't get any games to work, could be my roms though.

Bard - Can't launch epub. Black screen, then right back to file selection. .txt file is fine.

CCDoom - left shoulder button (strafe) is also mapped to swap weapon.

ASCII Portal - Was booting to black screen and freezing the system. Then it worked once. Tried it again and it gave me a black screen again. Now when I try to launch the game it just goes right back to gmenu2x. 

Meteoroid - Seems like the 3D graphics are messed up. Plays fine otherwise.

Noiz2sa - Can't actually start any of the levels no matter what button I press.

Quake - runs way to fast, and the buttons don't seem to be mapped correctly.

Handy - Doesn't load any games

FBA a320 - the ingame quick menu L+R+Start, is not displayed correctly (squished).

ScummVM - Not really a bug but the included version of ScummVM doesn't support FLAC audio. There is a version http://wiki.scummvm.org/index.php/Dingux that does, maybe it could be added? 

USB - This was a bug in the original firmware, probably nothing can be done. You need to restart the device every time you want to use USB to computer.

jaZzKCS, 2018-03-15:
> TV out
I fiddled around with this a little bit. It seems to be related to whatever you have set as signal in the GMenu settings (NTSC/PAL)? Because after FCEUX froze I tried again with setting to NTSC in GMenu and using a USA rom in NTSC mode in the emulator. Which worked. Once. then it crashed again. Maybe not related, after all.

> USB
Needs to be connected from the get go, indeed. Maybe a limitation?

steward-fu, 2018-03-16:
Fix for CCDoom:
https://github.com/steward-fu/gh_retrogame_binary

Please help to try it

jutleys, 2018-03-16:
Ok thanks is it the left shoulder button remapped fix because i think it was changing weapon.

jutleys, 2018-03-16:
Also Steward if you need another RS-97 for development funnyplaying.com jay provided to me for $29 very cheap if you need one let me know i can pass on your email to him.

steward-fu, 2018-03-16:
Fix CCDoom (disable left shoulder remapped issue):
https://github.com/steward-fu/gh_retrogame_binary

steward-fu, 2018-03-16:
FCEUX - all video scalers don't display correctly except original, and aspect.
https://github.com/steward-fu/gh_retrogame_binary

jaZzKCS, 2018-03-16:
FCEUX scaler fix tested and working! :+1:

jutleys, 2018-03-16:
CCDoom tested wont start on my device again after i copied and overwrite the files.

jaZzKCS, 2018-03-16:
CCDoom works for me when choosing the wads from the CZDoom directory. BUT the strafe button now seems to be the same as the fire button (which is semi-convenient as it turned out :P)

steward-fu, 2018-03-17:
@jaZzKCS, @jutleys 
remove /mnt/int_sd/.chocolate-doom/ and then overwritten it.
I have updated new one
Thanks

steward-fu, 2018-03-17:
Fixed Handy - Doesn't load any games issue.
Please help to test it

jutleys, 2018-03-17:
Still the same returns back to the roms list.

steward-fu, 2018-03-17:
kindly provide full rompath to me

jutleys, 2018-03-17:
/mnt/ext_sd/Atari Lynx

steward-fu, 2018-03-17:
do you have updated run.dge file ?

jutleys, 2018-03-17:
yes but what is the best way to download the fixed files i installed the
github on pc and then sync your changes then open in explorer and just copy
the files from there is this the corect way to get the latest files?

steward-fu, 2018-03-17:
syncing via git is ok.
I think it should be related rom issue, not emulator.
Can you provide this rom to me ?

jutleys, 2018-03-17:
It's the same romset unzipped I used on the beta handy and that worked
fine.But a can send you a rom file from it so you can test it.

jutleys, 2018-03-17:
I have changed my romset now going to test again and see if it works.

steward-fu, 2018-03-17:
Since beta version of handy has hang issue, so, I changed source to another github.

steward-fu, 2018-03-17:
ASCIIPortal: workaround: fix ogg library issue by convert ogg file to wave format
kindly help to test it again. thanks