steward-fu/gh_retrogame_emulator #128: Port request (Atarist Dcastaway) opendingux

jutleys, 2018-03-16:
http://dl.openhandhelds.org/cgi-bin/dingoo.cgi?0,0,0,0,72,535

tackro, 2018-03-17:
indeed, nice emu

steward-fu, 2018-03-17:
Ported to RetroGame.
kindly check it
https://github.com/steward-fu/gh_retrogame_binary

jutleys, 2018-03-17:
Same problem steward will only run via explorer and if i click on the file
dcastaway.

jutleys, 2018-03-17:
I looks like all the games i am adding will not start from gmenu2x only
with explorer not sure whats wrong but all the games run well thank you.

jutleys, 2018-03-18:
Ok tested again Reminiscence running perfect though this is a game not emulator handy has loud sound but works fine.Now dcastaway works great controls are fine many games run must be zipped or unzipped in the ST format to run just select game then run.You have trigger buttons that call the mouse and keyboar.The only thing i noticed intro sound on games is a bit messed up but in game sound is fine but great emulator.

steward-fu, 2018-03-18:
For handy emulator, is "menu = backlight" key in your test ? or "menu = L button" ?

jutleys, 2018-03-18:
i will check

jutleys, 2018-03-18:
yes i just tested menu = backlight will exit emulator and then menu = L button will bring up the menu. Sound is too loud.

steward-fu, 2018-03-18:
the version of "menu = backlight" -> sound ok ?

jutleys, 2018-03-18:
sound is ok but too loud.