steward-fu/gh_retrogame_emulator #129: [pcsx4all] R2 and L2 buttons unavailable due to changed key mapping

jaZzKCS, 2018-03-17:
On the original GCWZero version, the mapping for ingame Select, R2 and L2 were as follows:
Select: B+Select
R2: Select+R1
L2: Select+L1

In the version from cfw1.2 the mapping is as follows:
Select: Select
R2: N/A
L2: N/A

In our version the Select button of the device is actually working as the actual ingame Select button, which normally is B+Select, as the select button was only used as a alternator key, and to get the actual ingame Select you had to press B+Select, as it functions as an alternator instead of a real key.

Something got mixed up there, making pressing Select+L1/R1 not activating L2/R2.

jaZzKCS, 2018-03-17:
This is how the button layout should look like. The ones marked in fat are not working/different:

In-game key mappings
PSX button - GCW Zero button(s)
D-PAD       D-PAD
Triangle    X
Square     Y
Circle        A
Cross        B
Start           Start
**Select      Select + B** (is only "Select" instead)
L1            L
R1           R
**L2           Select + L
R2           Select + R** (both not working)
menu      Select+Start

steward-fu, 2018-04-04:
1. Turn on "Core settings \ Analog Arrow Keys"
2. SELECT + L/SELECT + R ->L2/R2

steward-fu, 2018-04-26:
Kindly close this case if issue fixed.

closed 2018-04-26 by jaZzKCS