steward-fu/gh_retrogame_emulator #135: Possible bug in battery level calculation

m1024x, 2018-03-20:
If you test this line of code with real values, the result is always incorrect and leads to battery level 5 (=100%):
https://github.com/steward-fu/gh_retrogame_emulator/blob/2d34ff1a76070a2ed9f9fd50080fcf52ec1bec7f/gmenu2x/src/gmenu2x.cpp#L2206

test input for empty battery: val=3000, min=3000, max=4500
```
level = 3000 / ((4500 - 3000) / 5)
=> 10, gets limited to 5 in next line
```

This code calculates the correct battery level:
```
long diff = val - min;
if (diff < 0) diff = 0;
level = diff / ((max - min) / 5);
```

steward-fu, 2018-03-20:
very thanks for your findings :)

I have updated it on binary github:
https://github.com/steward-fu/gh_retrogame_binary

kindly help to test it.
Thanks

m1024x, 2018-03-20:
Looks like it's working now.

As my battery is fully charged (4500mV) I manually set maxBattery from 4500 to 5000 and 5500 and the battery indicator was showing 65% and 45%, which looks ok.

After all your work this should be renamed to FuMenu2x :-)

jaZzKCS, 2018-03-20:
Testing now as well. Showing 10% although full, since I need to wait and let the values change on their own.

I copied over the executable to the external card by USB, edited the conf file with the embedded text editor, and afterwards copied over the executable with the embedded filemanager as the game partition is not available over USB ;)

steward-fu, 2018-03-20:
please full charge and full discharge

jaZzKCS, 2018-03-21:
Still not working quite correctly.
After the fix I now have three steps. 100%, 45%, and 30%. (And of course, low battery.)
But there is nothing displayed between 45 and 100. Seems to be very finicky in regards to the battery values.

m1024x, 2018-03-21:
Do your values in gmenu2x.conf look similar to mine? (minBattery=3260, maxBattery=4500)

jaZzKCS, 2018-03-22:
I guess. Mine shows minBattery=3356 and maxBattery=4500.

kaplanfx, 2018-03-28:
I can confirm the issue JaZzKCS is seeing, I see only 100% and 45% and then sometimes 10% if I catch it right before it dies.

Note I'm using an ebay 1300mah battery, I wonder if that is having an impact?

m1024x, 2018-03-29:
The level and percentage calculation in the code seems correct, but the maxBattery value looks a bit too high: From what I read about 3.7V batteries the maximum should be about 4200 (4.2V) instead of 4500 (4.5V). This means that the current voltage value (which is used to get the maximum) is somehow wrong.

**But the real problem could be this:**

The voltage drop is usually not linear over time! Imagine a curve that is more flat at the beginning and gets very steep at the end. That could be the reason for the percentage to jump and skip some values. The non-linear voltage drop needs to be mapped to a linear battery percentage. The same was done for the GP2X ([see code](https://github.com/steward-fu/gh_retrogame_emulator/blob/3d9259b3629ac603cdc7ffe591e475b1bdbc4d95/gmenu2x/src/gmenu2x.cpp#L2241)).

To get a working battery indicator we'd need:
- the possibility to log the current voltage value and timestamp via gmenu
- a few people who log the complete discharge process (timestamp + current voltage)
- replace the current linear/proportional calculation with a mapping similar to [this](https://github.com/steward-fu/gh_retrogame_emulator/blob/3d9259b3629ac603cdc7ffe591e475b1bdbc4d95/gmenu2x/src/gmenu2x.cpp#L2241)

PS: Another thing people should bear in mind: The percentage that is shown in gmenu is not the exact percentage. At first the level is calculated from the exact percentage. Then the level is mapped to fixed percentage steps (10/30/45/65/80/100). This means that you won't see 20% or 90%.