steward-fu/gh_retrogame_emulator #146: [Port Request] Tile World

mbaran5, 2018-04-25:
Was hoping you could port https://github.com/senquack/tileworld-for-handhelds to RS97. I attempted to, but it segfaults.

steward-fu, 2018-04-26:
miss data files.
could you kindly help to provide it to me ?

mbaran5, 2018-04-26:
Here’s a free pack: https://www.dropbox.com/s/r5mstmsfh8ygp9b/CCLP1.zip?dl=0

Here’s a site with a large archive of level packs
http://www.pillowpc2001.net/?i=1

When I tried to port the game myself it compiled OK but segfaulted trying to omit SDL

steward-fu, 2018-04-26:
porting complete:
https://github.com/steward-fu/gh_retrogame_binary
https://github.com/steward-fu/gh_retrogame_emulator

steward-fu, 2018-04-27:
kindly close this feature request if complete.
Thanks :)

closed 2018-04-27 by mbaran5