steward-fu/gh_retrogame_emulator #148: Cdogs

jutleys, 2018-05-05:
https://github.com/cxong/cdogs-sdl/wiki/Developer-Getting-Started:-GCW-Zero

this is the instructions for c-dogs for the gcw-zero.

steward-fu, 2018-05-05:
Unfortunately, this source code still use SDL2 library as the following logs:
I will take more effort on this.
Thanks

Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  build/cmake/FindSDL2_mixer.cmake:76 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:71 (find_package)

jutleys, 2018-05-05:
OK thanks steward i wanted to ask how to change the rootfs boot logo splash
screen?

steward-fu, 2018-05-05:
/usr/share/splash/splash.jpg

jutleys, 2018-05-05:
Yes I find that but won't let me change permission

steward-fu, 2018-05-05:
create run.dge file and then run it, then you can modify rootfs:

#!/bin/sh
mount -o remount,rw / /

jutleys, 2018-05-05:
i have done it.

steward-fu, 2018-05-05:
do you run this file in original firmware or gmenu2xnext ?

jutleys, 2018-05-05:
> gmenunext

But i never done like the below method i did a different way .

create run.dge file and then run it, then you can modify rootfs:

#!/bin/sh
mount -o remount,rw / /

steward-fu, 2018-05-05:
different way ?

jutleys, 2018-05-05:
Yes steps below.

rename rootfs.dat to rootfs.img
then write that .img to a sd card.
then put in a linux ubuntu and change the splash.jp file
then create a image file again from the edited rootfs
rename it back to rootfs.dat and it works fine rootfs is 512mb.

steward-fu, 2018-05-05:
is it possible to take picture when boot rs97 to me?

jutleys, 2018-05-05:
Steward that is the 2nd boot logo how can we change the first boot logo
planet one ? ;) this is very hard to figure out.

jutleys, 2018-05-05:
Someone i know has got mame4all working rs-97 but screen if funny can you
try to fix it here

steward-fu, 2018-05-05:
ha ha, I misunderstand boot logo, ha ha, 
planet logo is written in nand-loader(before uboot), there are 2 methods can fix it:
1. rebuild nand-loader from jigenic source code (but it is not for rs97, need to do something)
2. binary patch it

mame4all ?
is it possible to provide git or source to me ?

jutleys, 2018-05-05:
this guy sent to me he said he is working on some emulators he fixed the
key maping of the amiga emulator here.

steward-fu, 2018-05-05:
I have tried to port xmame (from gcw0) to rs97, but emulator speed is not good.
for mame4all, basically, it can be ported from a320 also.
but if this guy is under developing, I think he will do more better than me. :)

steward-fu, 2018-05-05:
facebook: Steward Fu

jutleys, 2018-05-05:
Can you search for me Palwinder Jutley and add as friend i cant find yours